officinerobotiche/uNAV.X #26: Suspected overflow problem with K_ang parameter very low

katodo, 2015-01-04:
My motor have 43,3:1 ratio and 400cpr encoder, with Walter RobotParamCalculator https://github.com/officinerobotiche/RobotParamCalculator I obtain this value:
* k_vel: 14510820.570854
* k_ang:  0.000091

but with this value PID tuning is impossible.
I'm testing  http://automatica.ing.unibs.it/mco/tco/3_pid/zn.html algoritm with Ki and Kd = 0 and changing only Kp parameter.
I run **rosrun serial_bridge drive_bridge_node** to keep ENABLED bridge
and with this rostopic comman I give velocity "step":
**rostopic pub -1 /robot/command/velocity geometry_msgs/Twist -- '[0.5, 0.0, 0.0]' '[0.0, 0.0, 0.0]'**
**rostopic pub -1 /robot/command/velocity geometry_msgs/Twist -- '[-0.5, 0.0, 0.0]' '[0.0, 0.0, 0.0]'**

this is the result:

>**Left**: {P: 0.34, I: 0.0, D: 0.0}
>**Right**: {P: 0.34, I: 0.0, D: 0.0}
[image]

>**Left**: {P: 0.33, I: 0.0, D: 0.0}
>**Right**: {P: 0.33, I: 0.0, D: 0.0}
[image]

>**Left**: {P: 0.32, I: 0.0, D: 0.0}
>**Right**: {P: 0.32, I: 0.0, D: 0.0}
[image]

>**Left**: {P: 0.30, I: 0.0, D: 0.0}
>**Right**: {P: 0.30, I: 0.0, D: 0.0}
[image]

*MUST BE NOTED DIFFERENCE FROM LEFT and RIGHT MEASURE and rapid changing of response from 0.33 and 0.32 Kp value*

If I restore original value of k_vel: 27925268.0  (it is approximately twice than my K_vel ) pid response can be ok, small chang of Kp perform small change in response, but persist a different response of right channel:
[image]

Finishing, with *k_ang:  0.00017453292093705386* ( original value ) I've similar response from 2 channel:
[image]

Now I open my Encoder to check configuration...

rbonghi, 2015-01-04:
@guiott Do you remember a similar problem with your encoders in Lino robot?

@Myzhar

rbonghi, 2015-01-04:
I think if we solve this problem we solve another issue #12

guiott, 2015-01-04:
Yes, I remember I’ve had some problems but I don’t remember the solution.
My problem was something different. Over a particular speed it starts randomly but often generating wrong behavior, the calculated speed was 0 and it requires some cycles to come back at the right speed. This caused some bad and dangerous noises.

I’ve looked on my source code for some comment about this but with no success.

> 
> @guiott <https://github.com/guiott> Do you remember a similar problem with your encoders in Lino robot?
> 
> @Myzhar <https://github.com/Myzhar>

___________________
http://www.guiott.com/

Myzhar, 2015-01-04:
I think this is an overflow issue. I faced similar problem tuning the pid for my robot with K_p very low.
Now, if I remember well, we are working with millirad, this takes to number too high for certain configurations.
Is it really necessary to work with millirad? I think that centirad should be enough... If not decirad. .. The pid for the dspic works in fractional, so millirad is really dangerous and unuseful.

Myzhar, 2015-01-07:
Parsing the code I noticed that only K_vel affects the behavior of the PID.
The file involved is "motors_PID.c" at lines 303 and 330 in the functions 
int MotorPIDL(void) amd int MotorPIDR(void).

Line 303: 
if (SIG_VELLtmp) motor_left.measure_vel = SIG_VELLtmp * (parameter_motor_left.k_vel / timePeriodLtmp);

Line 330:
if (SIG_VELRtmp) motor_right.measure_vel = SIG_VELRtmp * (parameter_motor_right.k_vel / timePeriodRtmp);

I'm going to test them...

Myzhar, 2015-01-10:
A possible fix is available in the new branch "feature/gain_scaling".
@katodo please test it and let me know if the behavior is better and how it works now

rbonghi, 2015-01-10:
My idea is to move **K_vel** in gain loop control. 
[image]

If you look the pitcture, we have an equivalence when **K_B = K_vel*K_pwm** and **K_A = 1**
We now using the first schema (A)

rbonghi, 2015-01-14:
@katodo Do you can try again your test?

katodo, 2015-01-14:
Not soon, my problem was PID tuning related and with actual utility I need
a lot of time and concentration. I would like to avoid making mistakes again

Cordiali saluti
                        Mauro Soligo

Katodo  di Mauro Soligo
Via Molino Nuovo, 31 35012 Camposampiero PD Italy
P.IVA 04161520285  C.F. SLGMRA80L18B563Y
Cell: +39 3462137146  FAX: +39 0498591588  e-mail mauro.soligo@katodo.com

rbonghi, 2015-04-09:
This issue is connected with #37

closed 2015-04-09 by rbonghi