officinerobotiche/uNAV.X #37: PID: the K_p parameter does not effect the contro

Myzhar, 2015-01-29:
Trying to solve issue #36 I faced a problem in the PID controller: the K_p parameter is not enough to control the system.

I explain better. K_p, K_i and K_d are "fractional" variables, so they vary in the range [-1, 1).

Myzhar, 2015-01-29:
Another "problem" is related to the combination of the three parameters.

According to page what is written at page 74 of this document:
http://ww1.microchip.com/downloads/en/DeviceDoc/16bit_Language_Tool_Libraries_51456c.pdf

K_p + K_i + K_d should always be minor than 1, otherwise the coefficient abcCoefficients[0] will saturate.
At the same way Kp + 2*Kd should always be minor than 1, otherwise the coefficient abcCoefficients[1] will saturate.

In the PID Tuner application I inserted the controls. But I noticed that neither @rbonghi, neither @guiott verify these constrains in you firmware...

rbonghi, 2015-05-13:
We have three different implementation:
http://blog.opticontrols.com/archives/124

Now we use the **Parallel Algorithm**, but you prefer **Noninteractive Algorithm**. @Myzhar

rbonghi, 2015-05-13:
This is a little guide about our PID controller:
http://www.nptel.ac.in/courses/108105063/pdf/L-14%28SS%29%28IAC%29%20%28%28EE%29NPTEL%29.pdf

closed 2016-10-02 by rbonghi