officinerobotiche/uNAV.X #58: Verify enable levels

Myzhar, 2015-11-19:
Using qt_simple_unav_console we noticed that setting enable to HIGH on the relative pin there is a value of 0V, while setting it to LOW there is a value of 3.3V.

Looking at qt_simple_unav_console and at orblibcpp it seems that the value of enable is passed correctly, so probably (but not for sure) there is something wrong in the firmware.

A double cross check is needed to understand at which level there is the inversion.

rbonghi, 2015-11-23:
Hi, when I'll finish to write my PhD-thesis I'll start to work again to this code. Now I'm in stand by! :disappointed:

Myzhar, 2015-11-23:
Don't worry... this is only a reminder 😉

rbonghi, 2016-10-01:
Hi @Myzhar I've check the code about your issue and the code work better.

If you speak about the Enable level in parameter in `motor_parameter_bridge_t`

``` c
typedef struct __attribute__ ((__packed__)) _motor_parameter_bridge {
    uint8_t enable;
    uint16_t pwm_dead_zone;
    uint16_t pwm_frequency;
    float volt_offset;
    float volt_gain;
    float current_offset;
    float current_gain;
} motor_parameter_bridge_t;
#define LNG_MOTOR_PARAMETER_BRIDGE sizeof(motor_parameter_bridge_t)
```
The definition is explained on this line: https://github.com/officinerobotiche/or_bus_c.X/blob/develop/includes/packet/frame_motor.h#L131

It's the default logic level to enable the H-bridge. Some H-bridges work in negative logic and when you set pin high disabled it.

I don't know if I answered for your issue. It's been a while :smile:

Myzhar, 2016-10-01:
(y)

closed 2016-10-02 by rbonghi